GuitarML/FunBox #3: Add a simple display & encoder to v4 revision

sejerpz, 2024-08-08:
What's your opinion about adding a simple display and an encoder to the hardware?

GuitarML, 2024-08-19:
I probably won't be adding a screen, I had a specific layout in mind when designing the Funbox. There is another great project that has a screen/encoder that I've used, so a screen in this project seems redundant to me, here is the other project:

https://github.com/bkshepherd/DaisySeedProjects

If someone did want to add a screen to FunBox, they'd probably need to either use a bigger enclosure, or design a board with smd components instead of through-hole.

sejerpz, 2024-08-20:
Thanks for the link, but soldering smd is too much for me :)

Funbox is such a great a simple design but I'd like to have more options to display info to the user beside just the two leds.

dccourt, 2025-05-29:
I found that it's easy to mount a similar screen to the bkshepherd one onto the outside of the FunBox case, just in front of the footswitches.  FunBox doesn't use the I2C1 pins, so I've been able to add successfully add the I2C screen variant to the bkshepherd code and have it running.  I don't think there's room for an encoder, but I don't think it's really needed, and I have in mind a software mod to use a knob when the footswitch is pressed, or similar, if I find that I do miss it.

I feel that a screen would add similar utility to FunBox as it does to the bkshepherd project - i.e. providing feedback when the controls are adjusted, which allows for the pedal to be self-documenting.

dccourt, 2025-08-31:
@da6353 Sure, see https://github.com/dccourt/FunBox/releases/tag/screen_v1 .

@GuitarML Sure, will take a while before I'm able to draft up a PCB.