GuitarML/FunBox #4: Funbox vers. 3.2 - Software pin mapping is wrong.

sejerpz, 2024-09-11:
After building the Funbox I tried some of your effects and I notice that some controls don't work at all.

So I tried the Template project and I notice the same.

Debugging the problem I see that some pin mapping are wrong in software, I see that daisy petal is used as a base.

The wrong mappings I found are: Footswitches 1 & 2, all the 3 switches
I didn't test the dip switch yet.

This is my working code:

`fs1.Init(seed::D25);`
`fs2.Init(seed::D26);`

`sw1a.Init(seed::D14);`
`sw1b.Init(seed::D13);`

`sw2a.Init(seed::D7);`
`sw2b.Init(seed::D10);`

`sw3a.Init(seed::D2);`
`sw3b.Init(seed::D4);`

I've also a question: why use the DaisyPetal as a foundation class? Why not creating a Funbox[v32] class with just the hardware and connections present?

GuitarML, 2024-09-16:
@sejerpz did you replace the daisy_petal .h and .cpp files in libdaisy with the ones in my Funbox/mod folder prior to building libdaisy? You’ll need to do that to get the correct switch mappings, I think the only ones I kept the same as the original daisy petal are the knobs.

sejerpz, 2024-09-23:
Oh, no, I didn't do that, how could I miss that clear notice. Sorry.

Anyway I wrote a funbox.h with all the code and used that instead.

If you are interested I can make a PR.

closed 2024-09-23 by sejerpz