GuitarML/FunBox #5: Mars models and IRs

xconverge, 2024-10-08:
Hey @keyth72, thanks for the project! I am wondering if you have any further documentation/comments about how the IRs and models were created for the mars project. (ir_data.h and all_model_data_gru9_4count.h)

I have everything more or less setup but want to use my own chosen IRs/models and understand that process completely

(I am pretty up to speed for the most part with the codebase and c++ so feel free to be as technical as you want)

xconverge, 2024-10-08:
I did enough reading in your other repositories to get a better idea of it

closed 2024-10-08 by xconverge

xconverge, 2024-10-08:
@GuitarML so I found a lot across your repos for the model as mentioned, but I can't find much about how the IRs were generated, I assume they are a .wav to txt conversion, anything to keep in mind for this?

GuitarML, 2024-10-15:
That’s exactly it, it’s the raw IR data converted to text and put into the .h header file. I also had shorten them a lot by trimming the end off, or else the Daisy seed couldn’t keep up with processing.