GuitarML/FunBox #7: Missing Footprints

CarFlip, 2024-10-29:
Hello.
I was updating the PCB from the schematic in KiCad, and it seems there are some missing footprints in the design. I’m not sure if these footprints were custom-made or from a library I haven’t downloaded. Could you let me know if they’re available in a specific library, or if there’s a recommended way to recreate or replace them?

Thanks for your help, and great work on the project

[image]

CarFlip, 2024-10-30:
Also i seems the schematics are still using the TL074 instead of the MCP6024

GuitarML, 2024-10-30:
@CarFlip Yes, I left out those footprint files because I believe where I got them were under a slightly different kind of license, so as a precaution I left them out of my repo. And yes I'll need to rename the quad op amp in the schematic. Thanks!

closed 2024-11-17 by CarFlip